bit-bots/bitbots_lowlevel #30: ROSException: publish() to a closed topic

NFiedler, 2019-10-01:
```
[ERROR] [1569921943.266307]: bad callback: <bound method PressureConverter.pressure_cb_right of <__main__.PressureConverter instance at 0x7feb02fcd320>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/bitbots/wolfgang_ws/src/bitbots_lowlevel/bitbots_ros_control/src/pressure_converter.py", line 134, in pressure_cb_right
    self.wrench_pubs[i+4].publish(self.wrenches[i+4])
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1041, in publish
    raise ROSException("publish() to a closed topic")
```

SammyRamone, 2019-11-07:
when did this bug happen? while starting or shutting down?

jgueldenstein, 2019-11-14:
since this did not occur again, we can close this issue

closed 2019-11-14 by jgueldenstein