bit-bots/bitbots_lowlevel #60: docu: segmentation fault when building docu

hendrikbrandt, 2020-09-17:
After activating the documentation for the bitbots_lowlevel package using the steps in [How-to-Doku](http://doku.bit-bots.de/meta/manual/tutorials/Doku-How-To.html), executing "catkin build bitbots_lowlevel --no-deps --make-args Documentation" to build it results in a segmentation fault. This only happens for this package. Not html files are generated.

Flova, 2020-09-18:
What segfaults? The compiler? Did you try to clean build?

ftsell, 2020-09-19:
I've already talked about this with hendrik. Sphinx-build segfaults and believe this to be cause by it importing our python modules but haven't narrowed it down yet.

SammyRamone, 2022-07-01:
looks outdated due to ROS2 switch

closed 2022-07-01 by SammyRamone