bit-bots/bitbots_lowlevel #73: ros control: improve parallelization

SammyRamone, 2020-11-20:
## Is your feature request related to a problem? Please describe.
The ros control node uses 4 threads for the 4 buses. Currently, the node is limited to a single CPU (see https://github.com/bit-bots/bitbots_lowlevel/pull/72). 

## Describe the solution you'd like
It could be improved by better handling the 4 threads and setting them on 4 different CPUs or something similar.

jaagut, 2023-08-09:
Is this still the case? Do we still want to do this?

SammyRamone, 2023-08-17:
> Is this still the case? Do we still want to do this?

Propably, it is not worth the effort. You can try it out and see if you can improve the cycle time, but I guess that the difference will be small.

closed 2023-08-17 by jaagut