bit-bots/bitbots_lowlevel #95: motor communication

RoboCary-cpu, 2022-03-26:
I have a problem when i try to communicate with robotis servo on jetson NX.
It turns to have great loss, when I send the message in the speed of 100HZ using dynaxiel SDK.
However, it works well in my computer.

Flova, 2022-03-26:
There was some USB setting that enabled some low latency mode iirc. It could be that. @SammyRamone knows Mord about this.

Flova, 2022-03-26:
http://doku.bit-bots.de/meta/manual/tutorials/lowlevel.html?highlight=usb

Flova, 2022-03-26:
There are some insights regarding this setup.

SammyRamone, 2022-03-27:
As @Flova pointed out above, did you set these [settings](http://doku.bit-bots.de/meta/manual/tutorials/lowlevel.html?highlight=usb#kernel)?
If you did and the problem still remains, you will need to use a logic analyzer to see if it is due to timeouts, corrupted packages or latency introduced by the jetson.

RoboCary-cpu, 2022-03-30:
OK，I have fixed the problem.
I use the rospy.spin to replace while true in the communication node, then it wroks!

closed 2022-03-30 by RoboCary-cpu